pvvx/THB2 #126: BOOT_TH05F_v20.hex cannot be flashed to TH05Y_V3.1/2024-10-18

redseasix, 2025-01-25:
I get the following error message:

_c:\>python3 --version
Python 3.13.1

c:\>pip3 --version
pip 24.3.1 from C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.13_3.13.496.0_x64__qbz5n2kfra8p0\Lib\site-packages\pip (python 3.13)

c:\>python3 c:\1\rdwr_phy62x2.py -p COM21 -e -r wh c:\1\BOOT_TH05F_v20.hex
=========================================================
PHY62x2/ST17H66B Utility version 20.12.24
---------------------------------------------------------
Connecting...
PHY62x2: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'001340c4 6222M005'
FlashID: 1340c4, size: 512 kbytes
PHY62x2 - connected Ok
Traceback (most recent call last):
  File "c:\1\rdwr_phy62x2.py", line 725, in <module>
    main()
    ~~~~^^
  File "c:\1\rdwr_phy62x2.py", line 650, in main
    hp = ParseHexFile(args.filename)
  File "c:\1\rdwr_phy62x2.py", line 56, in ParseHexFile
    taddr = (int(hexstr[3:7],16))
ValueError: invalid literal for int() with base 16: ''

c:\>_

What am I doing wrong?
My BOOT_TH05F_v20.hex has a sha256 hash of:

382a8a80bd6734250f490385b7f6ede80d8684c6005a4714881a2e02fdb95e7b

pvvx, 2025-01-25:
Files are downloaded from github using the "raw" link.

[image]

redseasix, 2025-01-25:
I have just done this again and I still get the same error
I once pulled a SHA256 hash 7799b9abf82615eb220cf2a53d1d36294aec32c23e4b6cd6c1fb437ed34d4611 *BOOT_TH05F_v20.hex

What else can I do differently?
Could you please compare my hash so that I really have the right file?

pvvx, 2025-01-25:
Name: BOOT_TH05F_v20.hex
Sise: 139341 bytes : 136 KiB
SHA256: d87ca6e7032702d47e6ba7b25dd4ac6402f6d95caa04c20186e7bcdb2e2c27f6

pvvx, 2025-01-25:
Github changes line ending characters in *.hex files.
For this reason, SHA256 will not match.

```
>git clone https://github.com/pvvx/THB2
>cd bin
>fc /b BOOT_TH05F_v20.hex E:\PHY62x2\git\THB2\bin\BOOT_TH05F_v20.hex
Сравнение файлов BOOT_TH05F_v20.hex и E:\PHY62X2\GIT\THB2\BIN\BOOT_TH05F_v20.HEX
0000000F: 0A 0D
00000010: 3A 0A
00000011: 31 3A
00000012: 30 31
00000013: 33 30
...
```

redseasix, 2025-01-25:
thanks for the info regarding sha256. I have now pulled the BOOT_TH05F_v20.hex from frok: /dromie/THB2 and that worked right away. to be on the safe side, I also got the file from here again, which didn't work. So back to the frok from dromie which then worked straight away.
With my modest knowledge I have not been able to find out why this is so.

Thanks for the quick answers and support 
now the TH05F works for me.

pvvx, 2025-01-25:
ok

closed 2025-01-25 by pvvx