pvvx/THB2 #129: Keychain fob with ST17H66C

b0brdobr, 2025-02-06:
I have a keychain fob with ST17H66C. I managed to get information from it with rdwr_phy62x2.py. The flash size is 256 KB. Does it mean it's too small and that chip is not compatible with the custom firmware? Please advise.
`$ sudo python3 rdwr_phy62x2.py --port /dev/ttyUSB0 -e -r i
=========================================================
PHY62x2/ST17H66B Utility version 27.01.25
---------------------------------------------------------
Connecting...
PHY62x2: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'00124085 6222M005'
FlashID: 124085, size: 256 kbytes
PHY62x2 - connected Ok
Flash Status: 0x00
Flash Serial Number: 4250304638333710
Send command 'reset' - ok`

pvvx, 2025-02-06:
https://github.com/pvvx/THB2/issues/94

b0brdobr, 2025-02-12:
I flashed it successfully and it appears to be working fine.

closed 2025-02-12 by b0brdobr