pvvx/THB2 #131: Feature request: report "manufacturer" and "model" to HA

szupi-ipuzs, 2025-02-16:
Hi, first of all - thanks for this great piece of software. Works great!
I don't know if you are aware of this nice HA integration called [Battery Notes](https://github.com/andrew-codechimp/HA-Battery-Notes), it helps a lot in remembering when the batteries were replaced and which exactly battery types are required for specific devices. For people like me, with many battery devices (and short memory) this is a life saver.
Now, I would like to use this integration with your firmware, the problem is the integration matches the device type by the "manufacturer" and "model" fields that are supposed to be reported when device is added to HA. Here's how it looks like for switchbot device:

[image]

And here's the matching entry in the Battery Notes library:

[image]

The problem is your firmware does not report these fields to HA, so there's no way to match it. Here's how BTH01 shows up:

[image]

Could you please make a change so that these fields are reported and be unique for a device type? This way we can add matching library entries to Battery Notes database and have this feature working in HA. Or maybe there already is a way to do it somehow?

Thanks!

pvvx, 2025-02-16:
This is not a device problem. You have **passive scanning** mode enabled.
All BLE devices announce their name when using **active scanning**.

szupi-ipuzs, 2025-02-16:
Thanks for explanation. And your "patience". I was not aware that I need to know BLE in order to use your firmware.

closed 2025-02-16 by szupi-ipuzs

pvvx, 2025-02-17:
> Thanks for explanation. And your "patience". I was not aware that I need to know BLE in order to use your firmware.

And I don't understand why you post screenshots and questions about third-party software here... 
This should be written to the authors of your program.

szupi-ipuzs, 2025-02-17:
Why so aggressive? I've just found this is a [known HA problem](https://github.com/Bluetooth-Devices/bthome-ble/issues/95), and I can see you took part in the discussion. You could've just pointed me there.

pvvx, 2025-02-17:
Because each question takes time to answer, instead of doing something additional in the firmware.