pvvx/THB2 #133: No Display and not BLE after Flash OK -> TH5 3.1

Hates1209, 2025-02-23:
I flash my Second BLE TH5F but after flashing, no Display an No Bluetooth

C:\Python\THB2-master> python3 rdwr_phy62x2.py -p COM5 -r we 0x10000 BOOT_TH05F_v21.hex
=========================================================
PHY62x2/ST17H66B/TG7100B Utility version 08.02.25
---------------------------------------------------------
Connecting...
PHY62x2/TG7100B: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'001340c4 6222M005'
FlashID: 1340c4, size: 512 kbytes
PHY6222 - connected Ok
Write Flash data 0x00010000 to 0x0003151c from file: BOOT_TH05F_v21.hex ...
Erase block 64k Flash at 0x00010000... ok
Erase block 64k Flash at 0x00020000... ok
Erase sector Flash at 0x00030000... ok
Erase sector Flash at 0x00031000... ok
Write 0x00002000 bytes to Flash at 0x00010000... ok
Write 0x00002000 bytes to Flash at 0x00012000... ok
Write 0x00002000 bytes to Flash at 0x00014000... ok
Write 0x00002000 bytes to Flash at 0x00016000... ok
Write 0x00002000 bytes to Flash at 0x00018000... ok
Write 0x00002000 bytes to Flash at 0x0001a000... ok
Write 0x00002000 bytes to Flash at 0x0001c000... ok
Write 0x00002000 bytes to Flash at 0x0001e000... ok
Write 0x00002000 bytes to Flash at 0x00020000... ok
Write 0x00002000 bytes to Flash at 0x00022000... ok
Write 0x00002000 bytes to Flash at 0x00024000... ok
Write 0x00002000 bytes to Flash at 0x00026000... ok
Write 0x00002000 bytes to Flash at 0x00028000... ok
Write 0x00002000 bytes to Flash at 0x0002a000... ok
Write 0x00002000 bytes to Flash at 0x0002c000... ok
Write 0x00002000 bytes to Flash at 0x0002e000... ok
Write 0x0000151c bytes to Flash at 0x00030000... ok
----------------------------------------------------------
Write Flash data 0x00010000 to 0x0003151c from file: BOOT_TH05F_v21.hex - ok.
Send command 'reset' - ok

pvvx, 2025-02-23:
https://github.com/pvvx/THB2?tab=readme-ov-file#fw-boot-and-ota

```
python rdwr_phy62x2.py -p COM5 -e wh BOOT_TH05F_v21.hex
```

Hates1209, 2025-02-23:
Thanks for fast reply! Sorry I over read this part..:(

I don’t understand what did it mean-> On thermometers with a screen, if the time display is not turned on, during start or restart the following is displayed for a short time:

"Bot 14" - FW Boot version 1.4

"APP 15" - FW APP version 1.5

Or is it enough to take your commando without the -r?

Hates1209, 2025-02-23:
Okay working, thanks!