pvvx/THB2 #136: TH05F_V1.2 no flashing

Tommaso6969, 2025-03-08:
I friend
I'm using the th05f_v1.2

[image]

and the adaptor below
  
[image]

my soldering are  VCC to VBAT+ ,  GND to VBAT- , RXD to RX1 , TXD to TX1 , RTS to REST_N

and this is the message error

[image]

Please someone can tell me where i'm wrong?

pvvx, 2025-03-08:
> my soldering are VCC to VBAT+ , GND to VBAT- , RXD to RX1 , TXD to TX1 , RTS to REST_N

TXD to RX1 , RXD to TX1

Tommaso6969, 2025-03-09:
Thanks for your time boss

closed 2025-03-09 by Tommaso6969