pvvx/THB2 #137: Question on support for Fahrenheit added in version 1.8

AuldTimer, 2025-03-11:
First of all, great project, working well! I see in version 1.8, the change was "Added display of temperature in degrees Fahrenheit".  I am using TH05F and looked through the source code and did not see any reference to this change.  I could be missing it.  It works well in Celsius, how does one change it to Fahrenheit?  Thanks

pvvx, 2025-03-12:
https://pvvx.github.io/THB2/web/PHY62x2BTHome.html -> Config

AuldTimer, 2025-03-12:
Thank you.  I flashed the BOOT and APP firmware via USB-COM and did not visit the config settings.  Thought it might be a button press combination...

glabifrons, 2025-03-18:
> https://pvvx.github.io/THB2/web/PHY62x2BTHome.html -> Config

How can we do this in the CLI (rdwr_phy62x2.py)?
I already have devices running v21 boot+app, but I'd like to change their display to Fahrenheit as well.

pvvx, 2025-03-18:
Switching to display degrees in Fahrenheit is done in https://pvvx.github.io/THB2/web/PHY62x2BTHome.html

glabifrons, 2025-03-18:
> Switching to display degrees in Fahrenheit is done in https://pvvx.github.io/THB2/web/PHY62x2BTHome.html

That's the only way?
The only bluetooth adapter that seems to work with mine (all 14 of them, I made a programming jig) is the one on my Home Assistant machine. As I understand it, you have to run the browser from the machine with the bluetooth connection, which isn't possible here since there's no way to get a desktop on the Home Assistant machine.
None of my other bluetooth devices can even see these, not even my phone, nor my Raspberry Pis.

I tried popping the back off one and resetting it while holding the button as I read that leave it in OTA mode. I was hoping that would make it visible to my other systems, but it didn't work. It did change from APP21 to BOT21 at boot-up and the little kid's face no longer is displayed... I don't know if that's intentional to indicate a particular mode or not.

glabifrons, 2025-03-20:
I forgot to mention: I'm using Ubuntu, so Chrome and Chromium cannot connect to the serial ports either.
This is why I initially flashed them with the CLI, and why I was hoping it might be possible to change their display config via the CLI.

aemseemann, 2025-03-23:
FYI: As an exercise in Python and Bluetooth, I implemented a small command line utility `thbtool.py`that should be able to do what you want. I've uploaded it to this branch https://github.com/aemseemann/THB2/tree/feature/cli-config-tool
The way to change settings is to first read the actual settings into a YAML file:
```
python thbtool.py -a <address> config save thbsettings.yaml
```
Then edit the *.yaml file according to your needs, e.g. change `display.temp_unit` to `fahrenheit`.
Afterwards, write it to the device:
```
python thbtool.py -a <address> config restore thbsettings.yaml
```
(Note: See `requirements_bt.txt` for Python dependencies)

glabifrons, 2025-03-25:
@aemseemann That looks awesome! Much appreciated! 
Unfortunately, I'm still stuck with only one machine that seems to be able to see these devices (which I find baffling). 
I was hoping to be able to do it over serial... I'll have to order another bluetooth transceiver, so it'll be a while before I can try this.

aemseemann, 2025-03-25:
If it's only the Fahrenheit setting you'd like to change then you could try flashing the latest firmware via serial. It allows toggling the display unit by pressing the button for 3 seconds.

glabifrons, 2025-03-25:
That's even better! Thank you very much!

AuldTimer, 2025-03-29:
I have TH05Y_V3.1 boards and have been using TH05F_vxx.bin .  I just updated to the new version 21 to try the new Fahrenheit switch change and now the display only shows 2 digits and a decimal point in the upper left quadrant and the rest of the display is blank.  I swapped the board with one with an older version of firmware and the issue followed the board / firmware so it is not an LCD problem.  I then flashed the original board / display back to version 20 and the entire display worked again.  Thoughts?

pvvx, 2025-03-29:
> Thoughts?

Fixed, update...

pvvx, 2025-03-30:
>  I am now just running the bleak scanner...

In Linux, to successfully receive BLE devices, you need to change the Bluez configuration files. You need to set the scan and connection durations to those described in the Bluetooth GIG standard (10 seconds)....