pvvx/THB2 #139: can't flash BTH01Y - it is not connected - please help :-)

Wasilij-Pupkin, 2025-03-24:
**1. English
2. Russian
3. Deutsch**

[image]

[image]

**1. === ENGLISH ===**
First of all, thank you very much for the work done to develop these sensors for the home assistant.
However, I encountered such a problem (see photos):

- USB-Serial adapter FT232RL FTDI USB 3.3 V 5.5V Type C
https://de.aliexpress.com/item/1005006486900917.html

- BTH01Y Version V.3.1 (2024-11-13) (attention NOT v.1.3 exactly V.3.1)

- soldered the contacts as in the picture and tried both options (TM, TX1, RX1 and before that REST_N, TX2, RX2)
- I tried to throw the yellow wire from TM to both CTS-pin and DTR-pin, nothing helps.

Thonny recognized the adapter on COM5

Then I did this command:

`python rdwr_phy62x2.py -p COM5 -b 115200 -r rf fullflash.bin`

```
===========================================================
PHY62x2/ST17H66B/TG7100B Utility version 08.02.25
---------------------------------------------------------
Connecting...
PHY62x2/TG7100B: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset error! Response: b''
Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!
```

And the same thing is shown if you don't connect the sensor board at all, but leave only the adapter.

I also tried shorting the TM to GND or VCC and nothing worked.

I don't know what to blame: the adapter, the sensor board or crooked hands :-(

Please help with whatever you can :-)

**2. === RUSSIAN ===**
Прежде всего огромное спасибо за проделанную работу по разработке данных сенсоров для домашнего ассистента. 
Однако я столкнулся с такой проблеммой (смотри фотки):

- USB-Serial адаптер FT232RL FTDI USB 3,3 V 5,5V Type C
https://de.aliexpress.com/item/1005006486900917.html

- BTH01Y Version V.3.1 (2024-11-13) (внимание НЕ v.1.3 именно V.3.1)

- припаял контакты как на картинке причем пробoвал оба варианта (ТМ, TХ1, RХ1 и до этого REST_N, TX2, RX2)
- желтый провод от ТМ я пробывал кидать и на CTS-pin и на DTR-pin ничего не помогает.

Thonny распознал адаптер на COM5

Дальше делал эту команду:

`python rdwr_phy62x2.py -p COM5 -b 115200 -r rf fullflash.bin`

```
=========================================================
PHY62x2/ST17H66B/TG7100B Utility version 08.02.25
---------------------------------------------------------
Connecting...
PHY62x2/TG7100B: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset error! Response: b''
Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!
```

Причем то же самое показывает если сенсорную плату вобще не подключать, а оставить только адаптер.

Замыкание ТМ на GND или на VCC я тоже пробывал и ничего не получилось.

Не знаю на что грешить: Адаптер, Сенсорная Платта или кривые руки :-(

Помогите кто чем может :-)

**3. === DEUTSCH ===**
Zunächst einmal vielen Dank für die Arbeit, die Sie in die Entwicklung dieser Sensoren für den Heimassistenten gesteckt haben.
Allerdings bin ich auf folgendes Problem gestoßen (siehe Fotos):

- USB-Seriell-Adapter FT232RL FTDI USB 3,3 V 5,5 V Typ C
https://de.aliexpress.com/item/1005006486900917.html

- BTH01Y Version V.3.1 (13.11.2024) (Achtung NICHT v.1.3 sondern wirklich V.3.1)

- die Kontakte hab ich wie auf dem Bild verlötet und beide Möglichkeiten ausprobiert (TM, TX1, RX1 und davor REST_N, TX2, RX2)
- Ich habe versucht, das gelbe Kabel von TM sowohl an den CTS-Pin als auch an den DTR-Pin anzuschließen, aber nichts hilft.

Thonny hat den Adapter an COM5 erkannt

Dann habe ich diesen Befehl ausgeführt:

`python rdwr_phy62x2.py -p COM5 -b 115200 -r rf fullflash.bin`

```
=========================================================
PHY62x2/ST17H66B/TG7100B Utility-Version 08.02.25
---------------------------------------------------------
Connecting...
PHY62x2/TG7100B: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset error! Response: b''
Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!
```

Dasselbe wird übrigens auch angezeigt, wenn die Sensorplatine gar nicht angeschlossen ist, sondern nur noch der Adapter vorhanden ist.

Ich habe auch versucht, das TM mit GND oder VCC kurzzuschließen, aber nichts hat funktioniert.

Ich weiß nicht, was die Schuld trägt: der Adapter, die Sensorplatte oder krumme Hände :-(

Bitte um jegliche Hilfe, bin für alles dankbar :-)

pvvx, 2025-03-24:
> припаял контакты как на картинке причем пробoвал оба варианта (ТМ, TХ1, RХ1 и до этого REST_N, TX2, RX2)

> Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!

TX->RX1, RX<-TX1, CTS->REST_N

Wasilij-Pupkin, 2025-03-24:
> > припаял контакты как на картинке причем пробoвал оба варианта (ТМ, TХ1, RХ1 и до этого REST_N, TX2, RX2)
> 
> > Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!
> 
> TX->RX1, RX<-TX1, RTS(or DTR)->REST_N
> 
> DTR [image] RTS [image]

Перепробовал все варианты. Жёлтый кабель от ТМ тыкал везде и в RTS и в DTR и в GND и в VCC, результат такой же как если только один адаптер подключен. 

Скорее всего плата не работает с этим адаптером или сам адаптер поломанный.

Wasilij-Pupkin, 2025-03-25:
> > припаял контакты как на картинке причем пробoвал оба варианта (ТМ, TХ1, RХ1 и до этого REST_N, TX2, RX2)
> 
> > Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!
> 
> TX->RX1, RX<-TX1, RTS(or DTR)->REST_N
> 
> DTR [image] RTS [image]

Sorry, I was DUMB, cuz i can't READ :-) - it works. 

At least i can read out fullflash.bin

This particular Issue can be closed - THANK YOU for fast answer :):):):):)

большое спасибо за указание на то, что я - дебил и не умею читать :-)

Поменял местами RX и TX и припаялся к REST_N

заработало - по крайней мере я счититываю fullflash.bin

Огромное СПАСИБО :)