pvvx/THB2 #141: Possible extra field in TH05F (TH05Y_V1.2,)

dirkx, 2025-03-31:
With firmware [BOOT_TH05F_v21.hex (https://github.com/pvvx/THB2/blob/a1f7727db3897fa4e07f494f18f6004f930c0e79/bin/BOOT_TH05F_v21.hex) I am seeing an 'extra' field that I am not expecting:

```
21:31:46.653 -> Device name: TH05F-FB382C
21:31:46.653 -> ServiceData: 1 block(s)
21:31:46.653 -> 	0 0xfcd2: (size=14) 40:00:62:01:59:02:06:08:03:4D:12:0C:53:0B
```

So decoding I get
```
 40: -- header
00:62 -- unkown
**01:59 extra/new field**
02:06:08 - temperature - 16 bits
03:4D:12 - humidity 16 bits float
0C:53:0B - battery - 8 bits
```

So the 0x40 is the normal BTHome header; we then get an odd 00:62 -- and then remainder is the usual data:

21:35:16.715 -> DEBUG: objID=0x01 => val=89.00, factor=1.000 unit=battery_percent
21:35:16.751 -> DEBUG: objID=0x02 => val=20.56, factor=0.010 unit=temperature
21:35:16.751 -> DEBUG: objID=0x03 => val=46.82, factor=0.010 unit=humidity
21:35:16.751 -> DEBUG: objID=0x0C => val=2.90, factor=0.001 unit=battery_voltage

pvvx, 2025-03-31:
https://bthome.io/format/ -> Packet id

dirkx, 2025-03-31:
Ok - so the fault must be in the parsing library I am using. Thanks. Will close & investigate that.

closed 2025-03-31 by dirkx

pvvx, 2025-03-31:
Specifies the measurement number. All packets with the same number are one measurement (e.g. the measurement at the current moment).
One BT4.0 packet may not contain all the information due to its size. As a result, the data is transmitted in several packets, but with one ID.

pvvx, 2025-03-31:
> Though I've not seen in practice that two packets with the same ID have different data for the TH05Y-V1.2 with v2.1:

There may not be a second block for TH05F yet. But what will be in the next version is still unknown...
In the current version there are two types of package - measurement data and events. But they have different IDs.
If add something, then you will need two data packets under one identifier, otherwise it will not fit into the encryption packet...

pvvx, 2025-03-31:
> This is with the NimBLE decoder (modified to also handle packet 00 and not error out).

I don't understand your efforts to correct some libs. The entire BTHome parsing code can be executed by the smallest MCU, since the entire program fits into a couple of hundred bytes of CPU code. The variable size table there is a bit larger...
https://github.com/pvvx/TLB2Z understands and parses several different formats on the fly and translates into Zigbee...