pvvx/THB2 #145: TH05y V3.1

bobybobybob, 2025-05-06:
c:\tempdig>python3 rdwr_phy62x2.py -p COM16 -b 115200 -e -r wh ./bin/BOOT_TH05F_v21.hex
=========================================================
PHY62x2/ST17H66B/TG7100B Utility version 08.02.25
---------------------------------------------------------
Connecting...
PHY62x2/TG7100B: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset error! Response: b''
Check connection TX->RX, RX<-TX, RTS->RESET, TM, and Chip Power!

**I had no big problems flashing some THB1   but with that TH05Y V3.1 it want work.**    

The PCB look like that from the TH05F Picture, the LCD like the TH05 V1.4

Bat+ to +3.3V
Bat- to GND
RX1 to TX
TX1 to RX
TN Pin to +3.3V

Reset to Reset    and try it with separate cable to GND + take of after flashprogramm is started. 
Also tried the RX2, TX2     and other Device CH341  and CP2112

pvvx, 2025-05-07:
> TN Pin to +3.3V

Why was "TM" connected to "+3.3V"?

`>rdwr_phy62x2.py -h`
```
usage: rdwr_phy62x2.py [-h] [--port PORT] [--baud BAUD] [--allerase] [--erase]
                       [--reset] [--start START] [--write WRITE] [--tm]
                       {wh,we,wf,er,ew,ea,rc,rf,i} ...

PHY62x2/ST17H66B/TG7100B Utility version 08.02.25

positional arguments:
  {wh,we,wf,er,ew,ea,rc,rf,i}
                        Run rdwr_phy62x2.py {command} -h for additional help
    wh                  Write hex file to Flash
    we                  Write bin file to Flash with sectors erases
    wf                  Write bin file to Flash without sectors erases
    er                  Erase Region (sectors) of Flash
    ew                  Erase Flash Work Area
    ea                  Erase All Flash (MAC, ChipID/IV)
    rc                  Read chip bus address to binary file
    rf                  Read all Flash
    i                   Chip Information

options:
  -h, --help            show this help message and exit
  --port, -p PORT       Serial port device
  --baud, -b BAUD       Set Port Baud rate (115200, 250000, 500000, 1000000)
  --allerase, -a        Pre-processing: All Chip Erase
  --erase, -e           Pre-processing: Erase Flash work area
  --reset, -r           Post-processing: Reset
  --start, -s START     Application start address for hex writer (default:
                        0x1fff1838)
  --write, -w WRITE     Flash starting address for hex writer (default:
                        0x00005000)
  --tm, -t              If pin TM is set "1"
```
`-t              If pin TM is set "1"`

bobybobybob, 2025-05-07:
Honestly, I have no idea. 

I read that user B4nd1 set the TM to 3.3V as a solution for flashing problems and was successful. He does not use the parameter -t in his example.  https://github.com/pvvx/THB2/issues/49#issuecomment-2117516820     Someone also wrote that this pin must be connected to VCC during the flash process and complained that this information was not included in the main instructions. And the stuff with the baud rate. 

Even with the data sheet of this chip I found nothing for ‘idiots’ how to flash the pure chip. Google Gemini gave me the pins that are usually used for TX1/2  RX1/2, and RESET. I measured the pads, they are connected right to the Chip. 

Uart0    TX = Pin P2      RX  = Pin P0       (Google Gemini say it is used for Logs)
Uart1    TX  =Pin P9      RX  = Pin P 10   ( it says it is used for Flashing)

+++++++++++++++++++

**I have now solved the problem in a somewhat stupid way. I removed the chip with hot air. I put the chip on a TH01 board, successfully flashed the TH05F with it and then transferred the chip to its original. I only bought two, so the effort was acceptable.** 

**Thank you for your comments**. Unfortunately, I only saw them too late to try them out. But maybe it will help someone else. 

[image]

closed 2025-05-07 by bobybobybob

pvvx, 2025-05-07:
> I read that user B4nd1 set the TM to 3.3V as a solution for flashing problems and was successful. 

Ask him why it doesn't work for you :)

For those who like to connect the "TM" pin to "+3.3V" to be original and not follow the instructions, the "-t" option was added.

bobybobybob, 2025-05-07:
I have a problem with English (use translator programm) and first connected it like last time with the THB1 and when that didn't work I googled it and came across the other hint. 

I know from these Tasmota sockets that the original manufacturers don't like reflashing and put things to make it more difficult or impossible, e.g. that solder bridges have to be made or the pads are not wired. The Chinese manufacturers are very creative in constantly bringing out different variants. 

As the saying goes: all's well that ends well.

pvvx, 2025-05-07:
In the end, the same steps are still required - reset chip before programming.
If "TM" is connected to +3.3V by default, the thermometer will not start. That's why the Chinese cannot connect "TM" to +3.3V. Only you can do this yourself.