pvvx/THB2 #146: Serial monitor not showing anything

klaus-liebler, 2025-05-15:
Hello, 
I would like to slightly extend the project's code for use in university teaching. My device is a TH05F.

 I was able to successfully

my USB-UART converter. Unfortunately, I am not receiving any data in my serial monitor (115200 8N1)

Do I need to make any further changes to get output, or am I using the wrong solder pad?

Thank you for this wonderful project and thank you in advance for your help!

[image]

Regards,

Klaus

denwilly, 2025-05-29:
I don't think you need the TX2.

gregd72002, 2026-09-26:
Have you tried to recompile with DEBUG_INFO=1?

Normally no LOG on serial will be generate unless enabled.