pvvx/THB2 #150: THB05F BLE not working

Zannys, 2025-08-25:
Hi, i have 3 this devices, 2 i can work perfectly, but when a flash the third display work ok, but i have no signal for BT, when i flash original FW i can find de device with nFR connect, any ideia?

[image]

pvvx, 2025-08-25:
BOOT is not written.
https://github.com/pvvx/THB2?tab=readme-ov-file#boot-and-app-firmware

pvvx, 2025-08-25:
Check the chip's "eFuse" using the extended [rdwr_phy62x2.py](https://github.com/pvvx/PHY62x2/blob/master/Utils/rdwr_phy62x2.py) program.
```
python rdwr_phy62x2.py -p COMx i
```

Zannys, 2025-08-27:
[image]

Zannys, 2025-08-27:
sorry, i didn't notice the is another file

[image]

Zannys, 2025-08-27:
i will try a older version to check, but is very strange, my other 2 sensors works fine, may be a full erase first?

Zannys, 2025-08-27:
just to check, this is normal output for flashing bootloader?

[image]

pvvx, 2025-08-28:
> just to check, this is normal output for flashing bootloader?

Yes.

> may be a full erase first?

The `-e` option clears Flash - "erase Flash work area". And this is displayed in the log.

```
PHY62x2/ST17H66B/TG7100B Utility version 01.03.25

positional arguments:
  {wh,we,wf,wc,er,ew,ea,rc,rf,i}
                        Run rdwr_phy62x2.py {command} -h for additional help
    wh                  Write hex file to Flash
    we                  Write bin file to Flash with sectors erases
    wf                  Write bin file to Flash without sectors erases
    wc                  Write bin file to chip bus address
    er                  Erase Region (sectors) of Flash
    ew                  Erase Flash Work Area
    ea                  Erase All Flash (MAC, ChipID/IV)
    rc                  Read chip bus address to binary file
    rf                  Read all Flash
    i                   Chip Information

options:
  -h, --help            show this help message and exit
  --port, -p PORT       Serial port device
  --baud, -b BAUD       Set Port Baud rate (115200, 250000, 500000, 1000000)
  --allerase, -a        Pre-processing: All Chip Erase
  --erase, -e           Pre-processing: Erase Flash work area
  --reset, -r           Post-processing: Reset
  --start, -s START     Application start address for hex writer (default:
                        0x1fff1838)
  --write, -w WRITE     Flash starting address for hex writer (default:
                        0x00005000)
  --tm, -t              If pin TM is set "1"
  --next, -n            Next commands (skip startup)
  --debug, -d           Debug
```

Clear all = `ea`  - "Erase All Flash (MAC, ChipID/IV)"

```
python rdwr_phy62x2.py -p COMx ea
```

bato3, 2025-12-16:
I thought I had this problem, but reading the documentation helped.

To enter boot mode:
1. Remove power
2. Hold down the button
3. Connect power

The screen will display "bot 21" and now you can connect to it

Normally, after flashing the entire firmware, only the "APP" version is displayed after a reset.

J0k3r2k1, 2026-08-19:
>  The spread returned to normal at supply voltages of 2.0-2.5V. At 3.0-3.3V, connection was impossible—BLE standards for connection timing accuracy were completely violated.

That was the solution for me. With about 2.2 V (from a bench power supply), the BLE signal was detected, and I was able to connect to the device. 🥳

pvvx, 2026-08-20:
> That was the solution for me. With about 2.2 V (from a bench power supply), the BLE signal was detected, and I was able to connect to the device. 🥳

It all depends on the chip series—specifically, whether a particular unit ended up in a defective batch...

And also on which dumpster the device manufacturer dug the chip out of.